TheSystemDevelopmentKit/rtl #4: Add support for Verilator

mkosunen, 2021-01-25:
This is a longer term development goal. To enabe student, and open source Verilog RTL development
, support for Verilator should be added https://www.veripool.org/wiki/verilator

Mainly this menas tconverting the Testbench handling featires of rtl-package to C++ including file IO and clock generation. Not difficult, but slightly time consuming task. Maybe a student assingment for some course, if no-one is willing to do that otherwise.

Roenski, 2022-04-28:
So, seems I am now the responsible for this. One thing that comes to mind right away: if we want to provide the "interactive mode" for compiling with Verilator, an Open-source waveform viewer is needed. The obvious choice here is GTKWave. Do you think we should add the support for it too?

mkosunen, 2022-04-28:
I think order should be 
1. Simulation with a 'testbench' that can be even written manually, but so that if follows the current naming convention tb_<entityname>.suffix . I assume that the testbench for verilator is in C. This is to clean up the compilation and execution path.
2. Automation for test bench generation similarly as it is currently for modelsim - Nice sugar abstracting away the boring part of testbench writing.
3. Interactive mode. GTKWave is fine.

Please wite the functionality inside 'verilator' module under rtl, and call the functionality from rtl class. LAter on we will isolate modelsim specific stuff to 'modelsim' package.

chiplet, 2022-04-28:
Verilator outputs C++ instead of plain C. I think we should use C++ language features where it makes sense to make our life easier, e.g. I'd imagine doing the file IO with the C++ stream abstraction is nicer to work with than using the plain POSIX interface. Same goes for memory management, since verilator testbenches usually also need to do head allocations.

This should be a good starting point for writing the testbench:
https://github.com/verilator/verilator/tree/master/examples/make_tracing_c

One thing to keep in mind is that verilator only supports synthesizable verilog features so it can't be used to run plain verilog testbenches which use non-synthesizable constructs like delayed assignments etc.

mkosunen, 2024-10-06:
Closed by #28

closed 2024-10-06 by mkosunen