TheSystemDevelopmentKit/rtl #36: Inverter testbench reads input and writes output at the same time

Anttitarkka, 2022-08-26:
Inverter testbench reads signal A and writes signal Z at posedge. Now it depends on the compiler which happens first and with Icarus Verilog the behavior is not deterministic therefore the IO files do not represent an inverter. Testbench should have some system that makes sure that input is read first.

Roenski, 2022-08-26:
@Anttitarkka since I think you don't have anything else to do, you could test whether it is caused by what we assume. So, try and come up with a way to test if it helps that the inputs are always read before outputs are written and see if it solves the problem. I think the first method to test is to have both `$fwrite` and `$fscan` commands in the same block. Easiest to write the testbench manually first, and if it works, then figure out what to do with the testbench generator.

Roenski, 2022-09-02:
For now, this was solved by changing the inverter example so that writing happens on falling clock edge. No changes to rtl needed.

closed 2022-09-02 by Roenski