TheSystemDevelopmentKit/rtl #94: Add capability to run simulation with or without optimization in QuestaSim

Roenski, 2024-07-09:
Currently the situation is as follows:

- Interactive simulations are always run with `-novopt` option, and there is no way to override it
- Non-interactive simulations are run with `-voptargs=+acc` option, which optimizes the design while maintaining visibility to all internal signals.

Some notes:
- In questasim, if no arguments regarding vopt are given, it runs the simulation with optimization enabled but not maintaining visibility to all internal signals.
- From my experience, having `-voptargs=+acc` makes the simulation almost as slow as with `-novopt`
- You can select which signals you want to keep by using `voptargs`. For example, to keep all top level signals, you can have `-voptargs="+acc=bcglnprst+ModuleName"`. Having all top level signals still creates a massive performance boost to simulations (we are talking stuff like 100x speed increase).

It would be nice if the following features were possible:
- Have the same defaults as now: interactive simulations run with -novopt and non-interactive with some kind of optimization
- Be able to run interactive simulations with optimization on
- Be able to run non-interactive simulations with optimization off?
- Be able to manually set voptargs. This is a powerful feature to manage trade-off between signal visibility and simulation time.

Some suggestions as to how to implement it:
1. Just use `vlogsimargs` defined in `sv` class, and let the user define them manually.
2. Add a property/function called `optimize` to `questasim` class. Here, you could, for example, provide the function with a list of modules whose signals you wish to keep.

The problem that occurs to me is how to add the default vopt behavior if the user does not want to define it manually? Here I explore the problem for the two suggestions introduced above:
1. If `vlogsimargs` is empty, then use default values? The problem becomes if the user defines a non-vopt-related argument, then it could change the vopt behavior even if the user did not specifically change it.
2. Here, we could define that if `optimize` function was not called, then we use the default behavior. However, problem arises if the user adds those functions manually to `vlogsimargs` then the behavior might be weird.

mkosunen, 2024-07-09:
Here's my two cents.

1. In any cade, those (visibility-speed) trade-off option should be defined in redefinable option.

2. Interactive simulations are for debugging, so defaulting to slowest, max-visibility is justified. This could be messaged with an info.

3. There could be 'preset library dictionary' per simulator  for some elementary trafe-offs, like nothing visible, only top-level visible  top-level+dut visible (if possible), and everything visible.

Roenski, 2024-07-09:
@mkosunen can you check the pull request linked above? I tested it with A-Core.

Roenski, 2024-07-09:
I believe the concept of optimizing a design does not exist in a lot of simulators, that's why I made it a questa-specific property.

closed 2024-07-19 by Roenski